slaclab/pabv_control #2: COM ports

bungernut, 2020-04-24:
We should look into something that tries a bunch or com ports or takes an argv, but leaves this as default.

https://github.com/slaclab/pabv_control/blob/a0b400357dda1c3e6de72b4bdc33c25a4d606cd0/python_client/client_dual.py#L27

bungernut, 2020-04-26:
Working on this in branch https://github.com/slaclab/pabv_control/tree/bung/gui_com_selector

mwittgen, 2020-04-29:
Should work if the vendor/product ID of the Arduinos used  are known by the GUI for COM lookup
`
import serial.tools.list_ports
for port in serial.tools.list_ports.comports():
    print(port.vid,port.pid,port.serial_number,port.manufacturer,port)
`
For an original UNO this returns:
9025 67 75433303631351D0E140 Arduino (www.arduino.cc) /dev/ttyACM0 - ttyACM0
Also no need to hard wire the /dev/tty on Linux
What about multiple Arduinos connected?

mwittgen, 2020-04-29:
Here's a list of a lot of known board VID/PID:
https://github.com/arduino/ArduinoCore-avr/blob/master/boards.txt

bungernut, 2020-04-29:
@mwittgen if you're comfortable working w/ pyqt (I'm not very) it might be nice to integrate this feature. Merging this branch is probably out of the question for other reasons, but this feature does work as I have it here I think.

mwittgen, 2020-04-30:
OK. Assign it to me.

slacrherbst, 2020-04-30:
The vendor ID approach is interesting. I use that at home to deal with multiple devices connected on usb to serial devices. It does assume the device type is know ahead of time which may not always be the case once we move to a usb to serial converter which have pretty generic device type IDs

slacrherbst, 2020-04-30:
It may also only work with linux based sytems

mwittgen, 2020-04-30:
@slacrherbst How does the final system look like? You will use the RX/TX pins for serial into a PC and no USB?

slacrherbst, 2020-04-30:
We will use standard RS232 and the user will supply a usb to serial adapter.

mwittgen, 2020-04-30:
OK. Going to add a MAX3232 to my test Uno and a USB-to-RSR32. What's reason not to use the USB directly? Packaging?

bungernut, 2020-04-30:
@mwittgen We wanted to be able to connect a laptop with the program running, and it would just receive serial input. So with an external USB-Serial adapter staying with the computer, the COM port stays the same and open. You can remove the connection from one device and plug straight into another and be up and running. 

If you think there is a better way, happy to hear it.

mwittgen, 2020-05-15:
Brian already implemented this. Next step: test and merge the branch add_serial.

mwittgen, 2020-06-02:
Closing. We are auto detecting the port.

closed 2020-06-02 by mwittgen