slaclab/pabv_control #4: Log reliability

bungernut, 2020-04-26:
Marty says the log is not bullet proof, not sure what that means.

marty-pmr, 2020-04-26:
Opening a file should be confirmed, as should closing. Opening twice seems to make a mess. The file itself seems pretty good, but should be well defined as to what it is.

mwittgen, 2020-05-15:
@marty-pmr @bungernut where do we stand on this?

bungernut, 2020-05-15:
I think only Ryan knows, maybe Marty does... probably not addressed though

marty-pmr, 2020-05-16:
I think this is still open. Not sure whether we need logging - perhaps for debugging???

mwittgen, 2020-05-16:
It should be removed then, the current version crashes. A true login feature might be better implemented on the arduino (not suggesting this). The client GUI as of now is stateless and adding file logging adds a dependency on the Win 10 laptop state. As of now the client could simply run from a USB stick and does not depend on any external  storage state and resources.

marty-pmr, 2020-05-16:
I have been using the log files to calibrate the spirometer. There could be ways of doing it without the log files…m

mwittgen, 2020-06-05:
Closing this for now. Has been fixed in devel.

closed 2020-06-05 by mwittgen