slaclab/pabv_control #6: memory?

bungernut, 2020-04-26:
I see lists are use to append data to. 
https://github.com/slaclab/pabv_control/blob/052e1d21fd8dd91c379374a21a47d0522e79b7ec/python_client/ambu_control.py#L99
Might make sense to use finite (numpy) arrays and let old data fall off.
Of course saving to file would still keep historical data.

bungernut, 2020-05-15:
I'm trying to implement a numpy FIFO class in ambu_contorl.py since that's where the data is. However this is not exactly straightforward in the abstraction to the GUI... I suspect @mwittgen knows both numpy and how the gui works enough to help?

bungernut, 2020-05-15:
well lets just say the class looks somethign like this
```
class npfifo:
    def __init__(self, num_parm, num_points):
        self._n = num_parm
        self._x = num_points
        self.A = numpy.zeros((self._n, self._x))
        self._i = 0
    
    def append(self, X):
        if len(X) != self._n:
            print("Wrong number of parameters to append, ignoring")
        # Move the data in the buffer
        self.A[:,:-1] = self.A[:,1:]
        # add the data to the end of the buffer
        self.A[:,-1] = X
        # increment number of 
        self._i += 1 
    
    def clear(self):
        self.A = 0.0
        self._i = 0
```

bungernut, 2020-05-15:
Adding something to get the data and we could check if there is enough data?
```
class npfifo:
    def __init__(self, num_parm, num_points):
        self._n = num_parm
        self._x = num_points
        self.A = numpy.zeros((self._n, self._x))
        self._i = 0
    
    def append(self, X):
        if len(X) != self._n:
            print("Wrong number of parameters to append, ignoring")
        # Move the data in the buffer
        self.A[:,:-1] = self.A[:,1:]
        # add the data to the end of the buffer
        self.A[:,-1] = X
        # increment number of 
        self._i += 1 
    
    def clear(self):
        self.A = 0.0
        self._i = 0
        
    def get_data(self):
        if self._i > 1:
            return self.A[:,-min(self._i, self._x):]
        else:
            return None
```

bungernut, 2020-05-15:
@mwittgen @slacrherbst We need to talk about this, in particular Matthias, if we're to keep flexible data stream this class seems incompatible with that goal.  Numpy in some sense becomes even less so than the current dict of lists... When is good time tomorrow morning?

bungernut, 2020-05-15:
@slacrherbst have a look at this 313778c89ca765070fd627bb1ee824a0e2e16c9e

mwittgen, 2020-05-17:
@bungernut can this be closed?

bungernut, 2020-05-17:
@mwittgen probably, assuming you're happy with the array structure.

closed 2020-05-17 by mwittgen