slaclab/pabv_control #10: GUI Reorganization

marty-pmr, 2020-05-09:
Move display and logging controls to a second page.
Make RR, Tinspiration, Trigger Pressure bigger.
Introduce PIP max parameter
Show measured peak PIP as number.
Show measured PEEP as number
Remove 3 state relay control from page and replace with:
         ON switch, large button outlined in green and filled with green when operating. On= cycling
         OFF switch Off = relay off. Off is large button outlined in red. First push causes green on light and red off light to to flash, and instruction to press off again to turn off ventilation. 2nd push turns off vent, green is off, red is steady.

mwittgen, 2020-05-15:
@slacrherbst Is this mostly done and can be closes?

bungernut, 2020-05-23:
We still need to figure out how to get PEEP per cycle. I see recent commit addresses PIPMax measurement 74dabacaf90a93a216b626a873a1eb936a20e299

PR #44 (2020-05-25): Additional bug fixes and features
closes #30 by suppressing debug messages.
closes #29 by suppressing debug messages.
closes #10  with new issue #43 created for on/off control change request.

Fixes vmax display bug.
closed 2020-06-03 by mwittgen