slaclab/pabv_control #15: Fix division by zeros

mwittgen, 2020-05-15:
For the pyinstaller bundle this will result in a program crash. 
@marty-pmr, @slacrherbst, @bungernut    Can division by zero handled by setting rates to zero or the display show Nan or blank?

bungernut, 2020-05-15:
More context would be good, permalink line?
I would say any of your options is fine

mwittgen, 2020-05-15:
https://github.com/slaclab/pabv_control/blob/f4e3325ee95e67712d412a8e6a295fc070bb6c22/python_client/ambu_control.py#L53
This one already fixed:
https://github.com/slaclab/pabv_control/blob/f4e3325ee95e67712d412a8e6a295fc070bb6c22/python_client/ambu_control.py#L184
I think there is not much else left at first glance.

mwittgen, 2020-05-15:
Fixed, pulled into devel branch.

closed 2020-05-15 by mwittgen