slaclab/pabv_control #30: Python Startup Error

marty-pmr, 2020-05-17:
This seems to happen every time:
base) C:\Users\mib\pabv_control\python_client>python client.py
Traceback (most recent call last):
  File "C:\Users\mib\pabv_control\python_client\ambu_control.py", line 126, in _handleSerial
    line = raw.decode('UTF-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x98 in position 0: invalid start byte
Got error 'utf-8' codec can't decode byte 0x98 in position 0: invalid start byte

mwittgen, 2020-05-17:
There should be no printout  from a traceback when catching this exception.

marty-pmr, 2020-05-17:
This happened at 1338 on Sunday:
(base) C:\Users\mib\pabv_control\python_client>python client.py
Traceback (most recent call last):
  File "C:\Users\mib\pabv_control\python_client\ambu_control.py", line 126, in _handleSerial
    line = raw.decode('UTF-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc2 in position 34: invalid continuation byte
Got error 'utf-8' codec can't decode byte 0xc2 in position 34: invalid continuation byte
Traceback (most recent call last):
  File "C:\Users\mib\pabv_control\python_client\ambu_control.py", line 126, in _handleSerial
    line = raw.decode('UTF-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc2 in position 34: invalid continuation byte
Got error 'utf-8' codec can't decode byte 0xc2 in position 34: invalid continuation byte
Traceback (most recent call last):
  File "C:\Users\mib\pabv_control\python_client\ambu_control.py", line 126, in _handleSerial
    line = raw.decode('UTF-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc2 in position 34: invalid continuation byte
Got error 'utf-8' codec can't decode byte 0xc2 in position 34: invalid continuation byte

PR #44 (2020-05-25): Additional bug fixes and features
closes #30 by suppressing debug messages.
closes #29 by suppressing debug messages.
closes #10  with new issue #43 created for on/off control change request.

Fixes vmax display bug.
closed 2020-06-02 by mwittgen