slaclab/pabv_control #31: Bug - Python stopped

marty-pmr, 2020-05-17:
[image]

marty-pmr, 2020-05-19:
Stopped again. Is this relevant?:
 File "C:\Users\mib\pabv_control\python_client\ambu_control.py", line 126, in _handleSerial
    line = raw.decode('UTF-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc5 in position 1: invalid continuation byte
Got error 'utf-8' codec can't decode byte 0xc5 in position 1: invalid continuation byte
Traceback (most recent call last):
  File "C:\Users\mib\pabv_control\python_client\ambu_control.py", line 126, in _handleSerial
    line = raw.decode('UTF-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc5 in position 1: invalid continuation byte
Got error 'utf-8' codec can't decode byte 0xc5 in position 1: invalid continuation byte
Traceback (most recent call last):
  File "C:\Users\mib\pabv_control\python_client\ambu_control.py", line 126, in _handleSerial
    line = raw.decode('UTF-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc5 in position 1: invalid continuation byte
Got error 'utf-8' codec can't decode byte 0xc5 in position 1: invalid continuation byte
Traceback (most recent call last):
  File "C:\Users\mib\pabv_control\python_client\ambu_control.py", line 126, in _handleSerial
    line = raw.decode('UTF-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x9b in position 12: invalid start byte
Got error 'utf-8' codec can't decode byte 0x9b in position 12: invalid start byte
Traceback (most recent call last):
  File "C:\Users\mib\pabv_control\python_client\ambu_control.py", line 126, in _handleSerial
    line = raw.decode('UTF-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x8f in position 0: invalid start byte
Got error 'utf-8' codec can't decode byte 0x8f in position 0: invalid start byte
Traceback (most recent call last):
  File "C:\Users\mib\pabv_control\python_client\ambu_control.py", line 126, in _handleSerial
    line = raw.decode('UTF-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x8f in position 0: invalid start byte
Got error 'utf-8' codec can't decode byte 0x8f in position 0: invalid start byte

mwittgen, 2020-06-05:
This particular issue is fixed.

closed 2020-06-05 by mwittgen