slaclab/pabv_control #51: Communication???

marty-pmr, 2020-05-24:
Found the time histories stuck. Python had no obvious problems. Could sort of stop/start cycling....Turn it off and a minute later it was going again.

mwittgen, 2020-05-24:
Seems a duplicate in some sense of closes issues#29 and #30. Instead of suppressing "debug" messages the should have a counter if packets are scrambled. It could be related to this issue.

marty-pmr, 2020-05-25:
Happened again this morning. Is there anything significant about a time of 88347.3 seconds Nano uptime?Is some 16 bit integer overflowing?

mwittgen, 2020-05-25:
It's a 32 bit millisecond counter and wraps around every 49 days.We really should keep statistics in the GUI on scrambled packets and try to guess  "package loss" by comparing excepted number of data packets, roughly GUI uptime compared to packets seen from the arduino.

mwittgen, 2020-06-05:
Communication was largely reworked.

closed 2020-06-05 by mwittgen