slaclab/pabv_control #67: GUI setting wrong parameters

bungernut, 2020-06-02:
https://github.com/slaclab/pabv_control/blob/154317fa86dcd02c7e7fa2e2bb6c78c3bd49fe46/python_client/ambu_control.py#L142

Looks like setter and parameter are mixed up on alot of these parmeters.

bungernut, 2020-06-02:
https://github.com/slaclab/pabv_control/blob/154317fa86dcd02c7e7fa2e2bb6c78c3bd49fe46/python_client/ambu_control.py#L130

bungernut, 2020-06-02:
https://github.com/slaclab/pabv_control/blob/154317fa86dcd02c7e7fa2e2bb6c78c3bd49fe46/python_client/ambu_control.py#L142

bungernut, 2020-06-02:
It's strange however that message.writeData is not returning anything, in fact it seems like it's not even returning None, it's like it's erroring silently and continuing on.

bungernut, 2020-06-02:
I think I figured out the problem, message.writeData() is expecting tuples, not numbers:
```
In [4]: m.writeData(m.PARAM_FLOAT, 0, (400.0), (5))
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-4-72cd003c73c7> in <module>
----> 1 m.writeData(m.PARAM_FLOAT, 0, (400.0), (5))

~\Documents\pabv_control\python_client\message.py in writeData(self, id, timestamp, float_data, int_data)
     66
     67     def writeData(self,id,timestamp,float_data,int_data):
---> 68         float_len=len(float_data)
     69         int_len=len(int_data)
     70         l=float_len+int_len*16

TypeError: object of type 'float' has no len()

In [5]: m.writeData(m.PARAM_FLOAT, 0, (400.0,), (5,))
Out[5]: b'xAAAAAARAADIQwUAAADmdQ==---'

In [6]:       
```

bungernut, 2020-06-02:
@mwittgen Can you confirm that you expected tuples in that class? If so I will fix these in the morning.

mwittgen, 2020-06-02:
I messed the brackets up. Fixed now, Closing

closed 2020-06-02 by mwittgen