slaclab/pabv_control #105: Make "adaptive" algorithm compatible with pressure-limited ventilation mode

kernal42, 2020-06-28:
When in pressure ventilation mode, the user sets the desired volume to something higher than the system is likely to reach (such as 700 mL), and the piston stroke ends when the max PIP pressure is reached.  With these settings, the volume correction factor drifts off with every stroke and becomes unreasonably large, providing poor system response if the user switches to a volume ventilation mode later.  

The correct behavior is probably to apply the adaptive algorithm and modify the volume offset only if the piston stroke was stopped by reaching the maximum volume.  In all other cases (inhalation time reached before max volume, or PIP reached before max volume) the adaptive algorithm should not be trying to learn.

bungernut, 2020-06-29:
I'm still confused by this. If we're in pressure mode, the "learning" should not matter. What it is supposed to do in pressure mode is squeeze until the time set by IT is reached. This is what the learning would do anyway.

kernal42, 2020-06-29:
There are two scenarios where I think this modification is relevant:
1) If the user switches from a pressure mode to a volume mode.  Long-running in the pressure mode will cause the correction factor to become very large.  When switching to a volume mode, now the calculation employs this unreasonably large correction factor and gives poor results until correction factor returns to baseline (presumably time proportional to amount of time running in pressure mode).  
2) The user has selected tight bounds on both volume and pressure.  If the system ever triggers on pressure, then the correction factor will receive a shock (as above) and the real volume trigger value will deviate.  

Note: (1) would also be solved by resetting the correction factor whenever use settings change.  I'm not sure that's a great solution though.

bungernut, 2020-06-29:
For one we know if the end result of the stroke comes from the Vmax. In that case only should we be applying the learning algorithm.  I think this fixes both cases?

There should perhaps be some range of values so that calculations do not run away causing some malloc or other C++ error.

bungernut, 2020-06-29:
Ok, i see that's exactly what you proposed. Are you sure that's not the way it's implemented?

bungernut, 2020-06-30:
Ryan, can you look into this?

marty-pmr, 2020-07-02:
I will try to define the two modes. m

slacrherbst, 2020-07-07:
Addressed in #107

closed 2020-07-07 by slacrherbst