slaclab/pabv_control #114: Change Volume Alarms

marty-pmr, 2020-07-14:
Based on EURS requirements, lets modify teh volume alarms:
In Volume mode, generate Low Volume (Medium Priority) at or below 80% VMax
Generate High Volume alarm at 120% of VMax.
As always, this in the On state after 5 breaths....
In Pressure, stay as is with the same Low Volume alarm.

slacrherbst, 2020-07-17:
Just to clarify. We need to add a volumeHigh alarm which did not exist before. Is this an alarm or warning.

Volume low goes from an alarm to a warning? In both modes?

marty-pmr, 2020-07-17:
Yes. In Volume mode, low and high are medium priority.
In pressure mode, volume below 250 ml is high priority.

bungernut, 2020-07-17:
This needs expanded to do this presumably:
https://github.com/slaclab/pabv_control/blob/d83d117002328973569dc3e3ef2438cf2efa231c/arduino/libraries/ambu_common/CycleControl.cpp#L180

PR #115 (2020-07-18): Update alarms
Closes #114
closed 2020-07-18 by slacrherbst