slaclab/pabv_control #133: Display Improvements

marty-pmr, 2020-08-18:
1. The encoder allows focusing on particular parameters by changing them to green. That is too hard to see. Perhaps the focused parameter could have its background changed and flash, or anything more obvious.

2. The primary alarm parameters should flash.

3. Decrease the encoder rotational sensitivity by 2 or 3.

bungernut, 2020-08-27:
@slacrherbst  is looking at 1 (perhaps just a change of color is possible) and 2 (we think making the corner square not used by any alarm flash is enough). 
We dont think 3 is super easy, if you have time Matthias maybe you can look at it? I am completely swamped by Snomass right now, maybe by the 30th I'm available.

bungernut, 2020-09-13:
I have tried just changing the color of the test to something less subtle.